jawi/ols #216: OSX DMG damaged?

chino, 2015-01-04:
When I try to run the bundle installed via DMG and from a local build I get:

```
“LogicSniffer” is damaged and can’t be opened. You should move it to the Trash.
```

I was able to run it from the command line using both the runner and with run.sh (after building it my self directly from git) but just having issues with the DMG launcher.

I'm on OSX 10.9.5.

jawi, 2015-01-05:
This is the gatekeeper functionality you're running in to. See [the FAQ](https://github.com/jawi/ols/wiki/FAQ) on the wiki for a solution...

chino, 2015-01-05:
Hm, I guess I've never seen that one before because it usually just says something like, "This is from the internet do you want to run it?".  Could this be because you didn't add an "Identified developer" to the Bundle?

jawi, 2015-01-06:
The only fix for this is to enroll in Apple's developer program, which is not without cost (99,= EUR/yr). I'm currently not in the position to invest that only to get rid of this message.

chino, 2015-01-06:
Np thanks.

ChrisHughes, 2016-10-13:
In case it helps anyone else, on OSX Sierra you can enter
`sudo spctl --master-disable`
to turn off Gatekeeper, as the 'Allow from Anywhere' option is no longer available to the end user.