jawi/ols #218: continuos capture mode?

chino, 2015-01-05:
Is real time capturing/rendering on the on the roadmap at all?

asdil12, 2015-01-06:
I don't think that USB is fast enough to do that.

The open logic sniffer uses (as far as I know) the FPGA to process the data and some other chip to cache it.
When a trigger is set up via ols software, the hardware is handling the trigger event and writes the datafeed into the cache until it is full.
Then the cache is sent to ols software via USB.

chino, 2015-01-06:
hm, I guess that makes sense.

I'm actually toying around with an Arduino project (https://github.com/gillham/logic_analyzer) that is only able to capture very small windows.

jawi, 2015-01-06:
The only valid use case I've ever heard for continuous acquisitions is to see anomalies in steady state situations. Obviously, this would only work if the DUT is running at low frequency.

chino, 2015-01-07:
ok thanks.

closed 2015-01-07 by chino