jawi/ols #219: Show device metadata problem

nikola-v, 2015-03-10:
Hi,

after connecting OBLS in OLS capture settings Show device metadata doesn't show anything, and Device type changes to empty.

Anyone has the same problem?

jogi997, 2015-04-06:
Hi there, noticed similar problem on my PC - it is either empty or states detection problems...

jawi, 2015-04-21:
This typically is due to an older firmware on the OBLS. Try upgrading to the latest firmware ("DemonCore")...

halt-hammerzeit, 2016-06-04:
This was indeed due to an old firmware even though I purchased the sniffer mid 2014.
I used these instructions for update (since they are an uneasy task to find)
http://samelimmat.blogspot.ru/2013/05/open-bench-logic-sniffer-with-os-x.html