jawi/ols #223: signal display blanks after scrolling at some level of zoom

dlarue, 2015-06-07:
when I scroll horizontally while analyzing a capture the signal display eventually go blank on the page. This only happens after a certain amount of zoom.  I've tried this on 0.9.8-SNAPSHOT and 0.9.7.2 and get the same results.

Running on Ubuntu 14.04 and java =java full version "1.7.0_79-b14"

jawi, 2015-06-08:
Do you have a capture file for me in which this happens? Is it only when zooming in, or also in other cases?

dlarue, 2015-06-08:
zooming is the only way I've found so far. I think I captured the project and linked the olp file. If it requires the csv or vcd file let me know. zooming in 5 times shows the blanking when you scroll to the end of the captured pulse train.

https://drive.google.com/file/d/0B6e_QEifuKzMSlpDb1FfZlYzTE0/view?usp=sharing