jawi/ols #228: 1-wire analyser works incorrect

aivs, 2015-11-04:
On [dangerousprototypes.com/forum](http://dangerousprototypes.com/forum/viewtopic.php?f=23&t=2767) I found, that users can successfully analyze 1-wire.
But for me 1-wire analyser not works correctly.
See screenshots:
[image]
*by arhi » Wed Sep 07, 2011 10:33 pm* - All OK

[image]
*by aivs » Tue Nov 03, 2015 4:49 pm* - Bad analyzed bytes (ols 0.9.7.2 with Arduino UNO)

As you can see, impulses are identical, but analyzed bytes different.
Can you check, is 1-wire analyzer works?

tudorp, 2016-09-18:
Dear aivs,

I did encounter very similar issues regarding onewire timing and parsing while working with DS1820 and another Dalas one wire memory within a dell lapto charger. Based on my investigation the problem is connected with over-simplified pulse tracing/parsing and with timing limits that are not close enough to the one wire protocol specifications (e.g. http://pdfserv.maximintegrated.com/en/an/AN937.pdf). I am working on a fix and preparing a pull request.

jawi, 2016-09-22:
@tudorp : it appears your changes caused a [test failure](https://drone.io/jawi/OLS-client/79) in the 1-wire decoder. Are you able to verify whether this is due to a faulty test or due to your changes?