jawi/ols #230: 0.9.7 SP2 requires JRE 6 on MAC OSX El Capitan

soswow, 2016-02-06:
Here is what I see when I start it:
[image]

I see it says 
```
<key>JVMVersion</key>
<string>1.6+</string>
```
But it doesn't help much seems like.

(I tried to change it manually to 1.8+. Didn't help.
I will follow the instruction and install legacy runtime just to make it work.

alexwhittemore, 2016-05-13:
I'm also curious about why this is the case, if the changelog indicated compatibility with 1.7 and 1.8 in a couple of updates.

andreas-manusm, 2016-10-19:
Same issue here, did you succeed and did you get it running on mac os x

jawi, 2016-10-23:
The problem is that the image isn't signed by a developer that is recognised by OSX: you need to pay some serious money to get a signing certificate from Apple.

There's a workaround described in the 2nd question of the OLS FAQ: https://github.com/jawi/ols/wiki/FAQ#Software. Basically, you've to temporarily disable the gatekeeper when running the OLS client for the first time. After this, you can enable the gatekeeper again...

sgentle, 2017-03-07:
I was able to get around this by just running Java directly:
```
cd /Applications/LogicSniffer.app/Contents/Resources/Java/
java -cp 'bin/org.apache.felix.main-4.4.1.jar:bin/runner-1.1.1.jar' -Xmx1024m -Dnl.lxtreme.ols.bundle.dir=plugins nl.lxtreme.ols.runner.Runner
```