jawi/ols #243: New project crash as in the issue #181

bourou01, 2018-04-25:
After clicking "new project", the following exception occurs:

[abdullatif@Host-001 ols-0.9.8-SNAPSHOT]$ ./run.sh
avr. 25, 2018 12:59:01 PM org.sump.device.logicsniffer.profile.DeviceProfile setProperties
INFOS: New device profile settings applied for Arduino Mega Logic Analyzer (AGLAM) ...
avr. 25, 2018 12:59:01 PM org.sump.device.logicsniffer.profile.DeviceProfile setProperties
INFOS: New device profile settings applied for Arduino Generic Logic Analyzer (AGLA) ...
avr. 25, 2018 12:59:01 PM org.sump.device.logicsniffer.profile.DeviceProfile setProperties
INFOS: New device profile settings applied for Original SUMP device (SUMP) ...
avr. 25, 2018 12:59:01 PM org.sump.device.logicsniffer.profile.DeviceProfile setProperties
INFOS: New device profile settings applied for USB Infrared Toy OLS mode (IRtoy) ...
avr. 25, 2018 12:59:01 PM org.sump.device.logicsniffer.profile.DeviceProfile setProperties
INFOS: New device profile settings applied for Logic Pirate (60 MHz) (LP60) ...
avr. 25, 2018 12:59:01 PM org.sump.device.logicsniffer.profile.DeviceProfile setProperties
INFOS: New device profile settings applied for Logic Shrimp (SHRIMP) ...
avr. 25, 2018 12:59:01 PM org.sump.device.logicsniffer.profile.DeviceProfile setProperties
INFOS: New device profile settings applied for Bus Pirate OLS mode (BP) ...
avr. 25, 2018 12:59:01 PM org.sump.device.logicsniffer.profile.DeviceProfile setProperties
INFOS: New device profile settings applied for Logic Pirate (40 MHz) (LP40) ...
avr. 25, 2018 12:59:01 PM org.sump.device.logicsniffer.profile.DeviceProfile setProperties
INFOS: New device profile settings applied for Open Bench Logic Sniffer (OLS) ...
JGoodies Looks: I have successfully installed the 'Sky Bluer' theme.
java.lang.NullPointerException
	at nl.lxtreme.ols.client.signaldisplay.SignalDiagramController$2.run(SignalDiagramController.java:508)
	at nl.lxtreme.ols.util.swing.SwingComponentUtils.invokeOnEDT(SwingComponentUtils.java:626)
	at nl.lxtreme.ols.client.signaldisplay.SignalDiagramController.setDataModel(SignalDiagramController.java:502)
	at nl.lxtreme.ols.client.MainFrame.updateViewData(MainFrame.java:983)
	at nl.lxtreme.ols.client.MainFrame$1.run(MainFrame.java:869)
	at nl.lxtreme.ols.util.swing.SwingComponentUtils.invokeOnEDT(SwingComponentUtils.java:626)
	at nl.lxtreme.ols.client.MainFrame.propertyChange(MainFrame.java:864)
	at java.beans.PropertyChangeSupport.fire(PropertyChangeSupport.java:335)
	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:327)
	at java.beans.PropertyChangeSupport.firePropertyChange(PropertyChangeSupport.java:263)
	at nl.lxtreme.ols.client.project.impl.ProjectManagerImpl.setProject(ProjectManagerImpl.java:534)
	at nl.lxtreme.ols.client.project.impl.ProjectManagerImpl.createNewProject(ProjectManagerImpl.java:83)
	at nl.lxtreme.ols.client.ClientController.createNewProject(ClientController.java:595)
	at nl.lxtreme.ols.client.action.NewProjectAction.actionPerformed(NewProjectAction.java:78)
	at javax.swing.AbstractButton.fireActionPerformed(AbstractButton.java:2022)
	at javax.swing.AbstractButton$Handler.actionPerformed(AbstractButton.java:2348)
	at javax.swing.DefaultButtonModel.fireActionPerformed(DefaultButtonModel.java:402)
	at javax.swing.DefaultButtonModel.setPressed(DefaultButtonModel.java:259)
	at javax.swing.AbstractButton.doClick(AbstractButton.java:376)
	at javax.swing.plaf.basic.BasicMenuItemUI.doClick(BasicMenuItemUI.java:842)
	at javax.swing.plaf.basic.BasicMenuItemUI$Handler.mouseReleased(BasicMenuItemUI.java:886)
	at java.awt.AWTEventMulticaster.mouseReleased(AWTEventMulticaster.java:289)
	at java.awt.Component.processMouseEvent(Component.java:6533)
	at javax.swing.JComponent.processMouseEvent(JComponent.java:3324)
	at java.awt.Component.processEvent(Component.java:6298)
	at java.awt.Container.processEvent(Container.java:2237)
	at java.awt.Component.dispatchEventImpl(Component.java:4889)
	at java.awt.Container.dispatchEventImpl(Container.java:2295)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.LightweightDispatcher.retargetMouseEvent(Container.java:4889)
	at java.awt.LightweightDispatcher.processMouseEvent(Container.java:4526)
	at java.awt.LightweightDispatcher.dispatchEvent(Container.java:4467)
	at java.awt.Container.dispatchEventImpl(Container.java:2281)
	at java.awt.Window.dispatchEventImpl(Window.java:2746)
	at java.awt.Component.dispatchEvent(Component.java:4711)
	at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:760)
	at java.awt.EventQueue.access$500(EventQueue.java:97)
	at java.awt.EventQueue$3.run(EventQueue.java:709)
	at java.awt.EventQueue$3.run(EventQueue.java:703)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:90)
	at java.awt.EventQueue$4.run(EventQueue.java:733)
	at java.awt.EventQueue$4.run(EventQueue.java:731)
	at java.security.AccessController.doPrivileged(Native Method)
	at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:80)
	at java.awt.EventQueue.dispatchEvent(EventQueue.java:730)
	at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:205)
	at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
	at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
	at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
	at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)

SleepyDeb, 2018-05-14:
Only if you have no data in your view :D

jawi, 2021-01-18:
Committed a fix in 14199bdb.

closed 2021-01-18 by jawi