jawi/ols #245: Build process not producing application

julian-weinert, 2019-02-04:
I'm trying to build the application on my machine but am not able to find the Mac application bundle.
The build process runs smoothly and `old.distribution/target/` gets created and the products are there. Unfortunately they don't contain an application bundle.

How can I build the OS X client from source?

partsdotpdf, 2019-03-24:
Use run.sh script as documented.

julian-weinert, 2019-03-25:
Where did you find this documented?
Yes, this can be used to run the Java application from the command line.

But I want the packaged .app application bundle, which should be created when building this on a mac using `mvn clean package`.

The build phases also run and I get indication for that in the log, but nothing really helpful:
```[INFO] --- maven-antrun-plugin:1.8:run (bundle) @ ols.distribution ---
[INFO] Executing tasks

main:

init:

create:

package:

sign-app:

create-dmg:

all:```

This ant plugin should create an app bundle and also a `.dmg` disk image. The responsible configuration can be found in `ols.distribution/src/main/assembly/ext/`

rhyde, 2020-05-06:
I can't get it to build with a modern jdk, has anyone else succeeded?

julian-weinert, 2020-05-10:
No, unfortunately not. I ended up using the raw java version :/

rhyde, 2020-05-10:
Any guidance on how to do that?  The build process is failing in the
javadoc, and I am not familiar with the build tools you are using so I was
unsure on the best way to clean it up.  (I am an experienced java
developer, but I don't want to duplicate efforts if there was a simpler way
to make it work again.)

Rich

jawi, 2020-05-16:
> Any guidance on how to do that? The build process is failing in the javadoc, and I am not familiar with the build tools you are using so I was unsure on the best way to clean it up. (I am an [...]

This problem you're describing (which you've also mailed me about) is due to the fact that you're trying to build it using a *Java Runtime* instead of a *Java Development Kit*. Ensure that your `JAVA_HOME` environment setting is correctly pointing to a JDK (in your case it should probably be set to `/usr/lib/jvm/java-8-openjdk-amd64/`. HtH

jawi, 2020-05-16:
> Where did you find this documented?
> Yes, this can be used to run the Java application from the command line.
>
> But I want the packaged .app application bundle, which should be created when building [...]

Its been a while since I've build the OLS client on OSX (I'm without a Mac since well over 2 years now), but I do remember that you need to pass an additional property to Maven for the DMG package to be created: `mvn -DcreateDMG=true package`, IIRC. Does that help?

julian-weinert, 2020-06-19:
Thanks @jawi for getting back to us.
I'm sorry I didn't answer earlier, must have missed it.

The issue is not the missing creation of the DMG. It does not even generate the app bundle which is later copied into the DMG.

Unfortunately I'm not a Java dev and totally alliterate when it comes to Maven and Java.
The build script does not seem to run any tasks after the main bundling.

Before I posted this issue I did dig pretty deep into MVN and this project, but still without real experience, so I really don't remember what I did find out (but it wasn't a lot).

I'm happy to provide you with any details you may need to give me a hint. I'm happy to do any testing or fiddling around on my dev machine, so any idea would be appreciated.

Thanks für the `-D` tip though, it unfortunately didn't help

julian-weinert, 2022-04-05:
Yay! I've just got somewhere. The latest commits from this year prompted me to try it again.
I couldn't build something with JDK 17, but setting it to 12 let me finish the build.

But again no osx products. I dug a little deeper once again and found a few things in the `app-bundle.xml` that didn't work.
For once I had to actually remove the `package.osx.skip` option, not only set it to false.

`createDMG` was needed to be set as environment variable, not passed to mvn. And since the complete app bundle creation depends on that to be set as well I never got anything.

So I now have an XML that actually creates an app bundle and optionally creates a disk image.
Do you want me to submit a pull request, or paste my changes here?