jawi/ols #250: ols incompatible with newer java versions?

jmoellers, 2020-12-09:
Hi,
I tried to run ols 9.7.2 on OpenSUSE Leap 15.2 and Kubuntu 20.04 but it refused to start complaining about "Unresolved constraints". I had to downgrade java to openjdk version 1.8.0 on both.

JanZerebecki, 2020-12-09:
```
ERROR: Bundle nl.lxtreme.ols.device.test [1] Error starting
file:/home/josef/lib/ols-0.9.7.2/plugins/test-1.0.0.jar
(org.osgi.framework.BundleException: Unresolved constraint in bundle
nl.lxtreme.ols.device.test [1]: Unable to resolve 1.0: missing
requirement [1.0] osgi.wiring.package;
(osgi.wiring.package=javax.swing.border))
org.osgi.framework.BundleException: Unresolved constraint in bundle
nl.lxtreme.ols.device.test [1]: Unable to resolve 1.0: missing
requirement [1.0] osgi.wiring.package;
(osgi.wiring.package=javax.swing.border)
        at
org.apache.felix.framework.Felix.resolveBundleRevision(Felix.java:4002)
        at org.apache.felix.framework.Felix.startBundle(Felix.java:2045)
        at
org.apache.felix.framework.Felix.setActiveStartLevel(Felix.java:1299)
        at
org.apache.felix.framework.FrameworkStartLevelImpl.run(FrameworkStartLevelImpl.java:304)
        at java.base/java.lang.Thread.run(Thread.java:834)
```

jawi, 2021-01-18:
I've committed a couple of changes that allows you to at least build and run the OLS client on Java 11 and newer (tested on Java 11 and 14)... @JanZerebecki and/or @jmoellers : could you verify this is working for you? If so I can cut a new release...

arc12, 2021-04-29:
Thanks for the fixes @jawi - I was able to complile and now run. Also a general thankyou for keeping the client maintained; a lot of the SUMP legacy is now abandoned or 404..., and I appreciate being able to continue to use my hardware!

One outstanding issue, which you may be aware of:
```
openjdk version "15.0.2" 2021-01-19
OpenJDK Runtime Environment (build 15.0.2+7-27)
OpenJDK 64-Bit Server VM (build 15.0.2+7-27, mixed mode, sharing)
```

On running (Windows):
```
WARNING: An illegal reflective access operation has occurred
WARNING: Illegal reflective access by org.apache.felix.framework.ext.ClassPathExtenderFactory$DefaultClassLoaderExtender (file:/C:/Users/Adam.Cooper/ols-0.9.8-SNAPSHOT/bin/org.apache.felix.main-5.6.10.jar) to method java.net.URLClassLoader.addURL(java.net.URL)
WARNING: Please consider reporting this to the maintainers of org.apache.felix.framework.ext.ClassPathExtenderFactory$DefaultClassLoaderExtender
WARNING: Use --illegal-access=warn to enable warnings of further illegal reflective access operations
WARNING: All illegal access operations will be denied in a future release
```

I can see this is ultimately a felix framework issue, but dont know enough to sniff out whether OLS is depending on an obsolete version, which is my hunch... as felix seems to be on version 7.

jmoellers, 2021-04-29:
I apologize for not having responded. I guess I thought "Hm, I'll do it tomorrow" and then there was too much work otherwise.
But, I second arc12's remark: I was able to complile and now run!
Platforms: Kubuntu 20.04, openSUSE Leap 15.2
Thanks!

ricksorensen, 2021-04-29:
Just a FYI on my experience compiling the latest version ... I got a good compile and running executable, but I had to go through some odd steps to get it. 

20210429: on Ubuntu 20.04 using system tools
```
ols$ java -version
openjdk version "11.0.11" 2021-04-20
OpenJDK Runtime Environment (build 11.0.11+9-Ubuntu-0ubuntu2.20.04)
OpenJDK 64-Bit Server VM (build 11.0.11+9-Ubuntu-0ubuntu2.20.04, mixed mode, sharing
ols$ mvn -version
Apache Maven 3.6.3
Maven home: /usr/share/maven
Java version: 11.0.11, vendor: Ubuntu, runtime: /usr/lib/jvm/java-11-openjdk-amd64
Default locale: en_US, platform encoding: UTF-8
OS name: "linux", version: "5.4.0-72-generic", arch: "amd64", family: "unix"
ols$ export JAVA_HOME=/usr

```
I had to run 
`ols$ mvn clean install` 
2 or more times to get a good compile and test results.  The first runs resulted in an error in the `testExecuteFail` test:
```
  <testcase name="testExecuteFail" classname="nl.lxtreme.ols.task.execution.BackgroundTaskExecutionServiceTest" time="0.019">
    <error message="Unexpected exception, expected&lt;java.util.concurrent.ExecutionException&gt; but was&lt;org.mockito.exceptions.verification.WantedButNotInvoked&gt;" type="java.lang.Exception"><![CDATA[java.lang.Exception: Unexpected exception, expected<java.util.concurrent.ExecutionException> but was<org.mockito.exceptions.verification.WantedButNotInvoked>
	at nl.lxtreme.ols.task.execution.BackgroundTaskExecutionServiceTest.testExecuteFail(BackgroundTaskExecutionServiceTest.java:228)
]]></error>
  </testcase>
```
Rerunning  the build 0 or more times resulted in a good finish.  Note that if I used  ` mvn install -rf :service.taskexecution `  with out any modifications to any files the build also completed correctly.
Going back to an earlier 2019 version and using Java 8 (1.8) I dd not have any issues.

Thanks for the great tool.

Enggmaug, 2021-06-07:
Hello, would you please be kind enough to provide a release that is compatible with the latest JAVA ?
That would be greatly appreciated.

cheers, 
G.

barbudor, 2023-01-22:
Thanks @Maff- 
but I can' get it to work correctly on my Lubuntu 22.10 with openjdk 18.0.2 (jdk, not jre)

I tried to build it myself but it fails 
``` 
[INFO] Results:
[ERROR] Errors: 
[ERROR]   ConfigFileRepositoryTest.testAddChangedResourceOk:77->createResource:157->createResource:162 » ExceptionInInitializer
[ERROR]   ConfigFileRepositoryTest.testAddResourceWithStoringOk:46->createResource:157->createResource:162 » NoClassDefFound
[ERROR]   ConfigFileRepositoryTest.testSaveRepoOk:126->createResource:157->createResource:162 » NoClassDefFound
[ERROR]   ConfiguratorImplTest.setUp:94 » NoClassDefFound Could not initialize class org...
[INFO] 
[ERROR] Tests run: 7, Failures: 0, Errors: 4, Skipped: 0
[INFO] 
[INFO] ------------------------------------------------------------------------
[INFO] Reactor Summary:
[INFO] 
[INFO] OLS Reactor POM 1.0.0 .............................. SUCCESS [  0.080 s]
[INFO] MyDoggy Library 1.5.0-SNAPSHOT ..................... SUCCESS [  0.887 s]
[INFO] OLS Framework API 1.0.5 ............................ SUCCESS [  3.366 s]
[INFO] OLS Utilities 1.1.0 ................................ SUCCESS [  3.287 s]
[INFO] OLS Testing Utilities 1.1.0 ........................ SUCCESS [  2.749 s]
[INFO] OLS Logging support 1.0.0 .......................... SUCCESS [  0.092 s]
[INFO] RxTx Serial library (RxTx version) 2.2.0-10 ........ SUCCESS [  0.230 s]
[INFO] Configuration Provisioning 1.0.0 ................... FAILURE [  0.667 s]
.....
[INFO] ------------------------------------------------------------------------
[INFO] BUILD FAILURE
[INFO] ------------------------------------------------------------------------
[INFO] Total time:  11.767 s
[INFO] Finished at: 2023-01-22T19:40:50+01:00
[INFO] ------------------------------------------------------------------------
[ERROR] Failed to execute goal org.apache.maven.plugins:maven-surefire-plugin:2.21.0:test (default-test) on project service.config: There are test failures.
``` 

Unfortunately I know nothing about Java

barbudor, 2023-07-28:
@gdi1942 it's normal that you can't push a branch to a repo that doesn't belong to you or you are not a collaborator
You should fork, create a branch in your fork and then send a PR to this repo
If jawi wants/can, he will merge
At least that will provide a branch, in your fork, from where we can pull the fix.

gdi1942, 2023-07-29:
Ah good idea.
https://github.com/jawi/ols/pull/257
https://github.com/gdi1942/ols/tree/Issue250